makerdiary/nrf52832-mdk #1: No available boards are connected

zet235, 2017-04-20:
- 使用osx
- 安裝完畢pyODC

但是卻提示No available boards are connected

```
$ pyocd-flashtool -l
No available boards are connected
```
請問還需多安裝什麼嗎

makerdiary, 2017-04-21:
@zet235 可以帮忙查看一下 pyOCD 的版本吗？ 

```
$ pyocd-flashtool --version

#我在 OSX 上的版本是 ：
0.8.1a1
```

另外之前在 Ubuntu 上遇到这个问题，发现是系统权限问题，可以使用管理员权限试试。

或者使用 python 的虚拟环境 [virtualenv](https://virtualenv.pypa.io/en/latest/) 进行尝试。

期待你的反馈！

zet235, 2017-04-21:
一樣是 0.8.1a1
另外我後來使用了`nrfjprog`工具
可以透過以下指令安裝

```
brew cask install nrf5x-command-line-tools
```

官方的example makefile  也是使用了這工具

```
# Flash the program
flash: $(OUTPUT_DIRECTORY)/nrf52832_xxaa.hex
    @echo Flashing: $<
    nrfjprog --program $< -f nrf52 --sectorerase
    nrfjprog --reset -f nrf52

# Flash softdevice
flash_softdevice:
    @echo Flashing: s132_nrf52_3.0.0_softdevice.hex
    nrfjprog --program $(SDK_ROOT)/components/softdevice/s132/hex/s132_nrf52_3.0.0_softdevice.hex -f nrf52 --sectorerase
    nrfjprog --reset -f nrf52

erase:
    nrfjprog --eraseall -f nrf52
```

makerdiary, 2017-11-07:
@zet235 如果没有问题了，该issue就暂时关掉了，感谢反馈～

closed 2017-11-07 by makerdiary