makerdiary/nrf52832-mdk #11: Board lists as unsupported

jefffhaynes, 2018-03-11:
Running firmware 245, pyocd-flashtool -l prints:

WARNING:root:Unsupported board found 1024
0 => Unknown Board [cortex_m] boardId => 102400000f00690c00000000000000000000000097969902

makerdiary, 2018-03-12:
@jefffhaynes It can be used normally. I‘m trying to apply the board ID for this board from Arm Mbed. 
You can add locally manufacturers id and platform name for nRF52832-MDK and see which boards are connected with `mbed detect`:
```
$ mbedls --mock=1024:nRF52832-MDK
```

jefffhaynes, 2018-03-12:
I got it working, albeit somewhat unstably

―
If you want to build a ship, don't drum up people to collect wood and don't assign them tasks and work, but rather teach them to long for the endless immensity of the sea.

Antoine de Saint-Exupery

closed 2018-05-24 by caizelin