makerdiary/nrf52832-mdk #12: Source code

ntavish, 2018-05-14:
Hi, I don't see the code pushed in https://github.com/makerdiary/DAPLink is it somewhere else?

caizelin, 2018-05-24:
@ntavish, the daplink code is ready. I will push the code as soon as possible

Django0, 2021-12-12:
Its December 2021 now, and I still can't find the code. Please point me to it. Thanks.

caizelin, 2021-12-13:
> Its December 2021 now, and I still can't find the code. Please point me to it. Thanks.

Hi,
The debugger on nrf52832-mdk is the same as Pitaya-Link. Just checkout this target.
https://github.com/makerdiary/DAPLink

Django0, 2021-12-13:
Thank you for the link.
I was following the procedure here: [https://wiki.makerdiary.com/nrf52832-mdk/daplink/#compile-your-project](https://wiki.makerdiary.com/nrf52832-mdk/daplink/#compile-your-project)
 
And I could not locate the lpc11u35_nrf52832_mdk_if.uvproj as shown in the above documentation.

I am new to working with Keil. Kindly help.

caizelin, 2021-12-13:
Hi,
Just use `lpc11u35_pitaya_link_if` instead. The project file will be generated by:
```
progen generate -t uvision
```