makerdiary/nrf52832-mdk #13: Nearby eddystone different output

Exi16, 2018-09-30:
Good afternoon, i have a problem, when i upload the default ".hex" file to the Nordic it works as beacon nearby, but when i compile the project "ble_app_eddystone" and i upload this new ".hex" file it is not recognized as beacon nearby. 

The default ".hex" file isn't supposed to come from this project?

Or, how can i configure this project in order to get the ".hex" file works as beacon nearby when i upload it to Nordic?

caizelin, 2018-10-04:
@Exi16 which mobile app are you using to discover the beacon?

Exi16, 2018-10-04:
Good afternoon actually I am using the following app. Beacon tools, and google nearby. Is the one recommended In your documentation.