makerdiary/nrf52832-mdk #16: Is it possible to use the MDK as a programmer?

Bhav97, 2019-03-28:
I have a custom nrf52 board from sparkfun and I wanted to know if it's possible to use [something like this](https://devzone.nordicsemi.com/f/nordic-q-a/20536/programming-external-custom-nrf52832-board-using-nrf52-dk) to program the custom board but use the MDK instead of the DK. 

I use pyOCD for flashing the MDK

caizelin, 2019-03-28:
@Bhav97 It is possible to program a custom nrf52 board using the mdk board. Pls check this discussion: https://community.makerdiary.com/t/flashing-a-nrf52840-dongle-with-a-modified-nrf52840-mdk/236/3. The custom firmware should work on nRF52832-MDK, too.

Bhav97, 2019-03-28:
@caizelin, Thanks for the quick reply. I'm gonna try it out right now. However before I start, do you know which firmware I should use to revert my MDK back to it's original firmware. 

Also do you mind sharing any resources on how the custom firmware was generated? It looks really interesting.

Thanks

Bhav97, 2019-03-28:
So I didn't wait for a reply and tried uploading the custom DAPLINK firmware, cp and sync ran successfully. Removed drive to boot it normally and it's not being detected. I mean absolutely zero `dmesg` logs, no entry in `lsusb`.

Plugged it back in holding the reset button and it shows up in bootloader mode (CRP DISABLD). So I tried copying a file from firmware/daplink and it doesn't work again. Tried it with all firmwares.

Any ideas on how to resolve this? I'm stuck in the bootloader mode, with no response from the device when booting normally(?)

caizelin, 2019-03-28:
> @caizelin, Thanks for the quick reply. I'm gonna try it out right now. However before I start, do you know which firmware I should use to revert my MDK back to it's original firmware.
>
> Also do you [...]

The original firmware is located here: https://github.com/makerdiary/nrf52832-mdk/tree/master/firmware/daplink.
Here is the tutorial about how to build daplink firmware: https://wiki.makerdiary.com/nrf52832-mdk/daplink/

closed 2019-03-28 by Bhav97