makerdiary/nrf52832-mdk #23: LPC appears to consume 17mA on average

Django0, 2021-12-12:
Hi, 
I checked my code with the official nRF52832 dev board from Nordic and then tried it to run on nRF52832-mdk. It works great. Thanks for the USB support.
However, I noticed that the LPC11U35 and other non-Nordic related components on the board appear to consume 17mA when doing nothing i.e. idle period.
Is it possible to lower the power consumption of the LPC11U35 or shutdown down the LPC11U35 after say 10 seconds after bootup and keep the nRF52832 running? Perhaps, can you point me to the source code for LPC11U35 so that I can perform these changes.

Looking forward to your reponse.

Thanks,
Tilak

caizelin, 2021-12-13:
@Django0 You can just supply the power via `VIN` pin. USB will power LPC11U35 and nRF52832, but `VIN` only powers nRF52832.

Django0, 2021-12-13:
Thank you very much for this valuable input. I will close this issue now. I will reach out if I run into problems.

closed 2021-12-13 by Django0